prusa3d/PrusaSlicer-settings #147: Remove or change "M221" commands from default Prusa printer profiles, and update "Extrusion Multiplier" value in default Prusa Filament profiles to compensate for them

cloud30000, 2022-01-08:
### Version
Version 2.4.0+win64 (Technically Profiles version (PrusaSlicer-settings) 1b34120278c658cea379cd1f795789e4d17a79cc, which is independent of the PrusaSlicer release version)

### Operating system type + version
Win 64

### 3D printer brand / version + firmware version (if known)
MK3S+

### Behavior
The Prusa printers have been determined long ago to over-extrude by Joseph, and as a means of mitigating the issue, the extrusion G-Code command "M221 S95" was added to the start code in the default Prusa Printer profiles.

However, custom G-Code in general (from start code, end code, before layer, after layer, as well as other entries) is not well handled by PrusaSliucer when all calculations are made and can lead to inaccuracies (such as in used filament, and when the max volumetric speed throttles print speed but end up 5% below the set volumetric limit after accounting for the M221 S95 value).

This request is to take advantage of the "Extrusion Multiplier" value included in all Filament profiles, which is more robust better utilized in all other Slicer calculation, instead of relying on the Marlin firmware to handle extrusion correction.

As such, the "M221" entries should be removed from the default Prusa Printer profile and instead be added to the Prusa Filament profiles.

This request should probably go in the "Prusa-settings" branch, but being a first-party printer and first-party filaments, I thought it better to go here for the appropriate consideration by Prusa's staff.

In the limited situations where a value other than "95" has been used, an update can be made to still take advantage of it.
For instance, "M221 S{if layer_height<0.075}100{else}95{endif}" can become "{if layer_height<0.075}M221 S105{endif}".

rtyr, 2022-01-08:
I don't disagree, but it would break all existing custom filament presets people have and it would also make all filament profiles incompatible with MK2 family printers.

cloud30000, 2022-01-08:
I can absolutely see that argument; and it is the risk and drawback of combining default profiles with (branched) custom profiles; if you update the default profiles but don't merge in the changes for the branched profiles, you will have issues. Unfortunately, this line of reasoning can lead to zero future progress once backwards compatibility blocks any additional improvements.

As far as the change breaking the default profiles for the MK2, that is a very simple solution; branch the filament profiles based on the printer, with the MK3S+ getting one extrusion multiplier and either setting the "Compatible printers" value or changing the "printer_notes=~/.*PRINTER_MODEL_MK(2.5|3).*/" condition to be "printer_notes=~/.*PRINTER_MODEL_MK2*/" for MK2 and "printer_notes=~/.*PRINTER_MODEL_MK3.*/" for MK3

cloud30000, 2022-01-08:
As a messy compromise, a note can be added to the new default Filament profiles with the string "Updated_to_remove_M221", and instead of removing the existing M221 statements, they can be amended with "{if filament_notes=~/.Updated_to_remove_M221./}M211...{endif}"

I find this approach a bit cumbersome and sloppy, but it will maintain backwards compatibility with custom filament profiles.

bubnikv, 2022-01-08:
We work on internal G-code post-processor, you will be able to delete these
lines using your Print profile.

cloud30000, 2022-01-08:
> We work on internal G-code post-processor, you will be able to delete these lines using your Print profile

Who works on the default Prusa Printer Profiles such as the MK3S+?

rtyr, 2022-01-08:
> As far as the change breaking the default profiles for the MK2, that is a very simple solution.

Creating and maintaining hundreds of duplicate filament profiles is not as simple as it may seem. But yes, it's one of the possible options which may be needed anyway because of other things, like filament overrides.

cloud30000, 2022-01-08:
> Creating and maintaining hundreds of duplicate filament profiles is not as simple as it may seem. But yes, it's one of the possible options which may be needed anyway because of other things, like [...]

I never claimed it would be quick or easy, just simple (albeit requiring a lot of work to implement and maintain as independent profiles).

The more complicated solution, which would negate the need to independently maintain every profile separately, is an update to the profile parser such that a profile ",ini" file can point/reference to a different existing profile, and then only specify the fields that will be over-ridden while using the values from the referenced file for all other fields.

bubnikv, 2022-01-08:
Profile inheritance is implemented since the beginning of our system
profiles, however it is currently limited to links inside a single vendor
ini file only. We plan to implement common shared profiles, however it is
highly non-trivial to implement if the remote updates are considered.

rtyr, 2023-11-21:
M221 S95 is no longer used with current printer models. We will not change it for legacy printers.

closed 2023-11-21 by rtyr