prusa3d/PrusaSlicer-settings #148: Fillamentum filament profile for DeltiQ 2 is incorrect

parhelia, 2022-02-03:
### Version
Version 2.4.1. beta 1 x64

### Operating system type + version
mac OS Monterey 12.2 (21D49)

### 3D printer brand / version + firmware version (if known)
TriLAB DeltiQ 2

### Behavior
Standard filament profile for Fillamentum on DeltiQ 2 is incorrect.
Cost is showing 750 money/kg, which would be insanely expensive. Standard PLA color should be about 22€ at the time of this writing.
Spool weight is showing 0, where it should be 230g for a standard 750g spool.

[image]

rtyr, 2022-02-03:
The cost is not in EUR, it is in CZK in this case.

It is not mandatory to use spool weights, but sure I guess it can be added with new TriLab bundle update.

@mat100 FYI

rtyr, 2022-03-25:
I am closing this. The price is in CZK and I notified maintainer of the Trilab bundle, so spool weight may be added with some future update.
We plan to improve filament cost setting in the future, so users can add custom value even for system profiles.

closed 2022-03-25 by rtyr