prusa3d/PrusaSlicer-settings #152: [Profile] Ultimaker

fleutot, 2022-03-09:
I just started using the Ultimaker

foreachthing, 2022-03-09:
Well... this is the problem with how PrusaSlicer starts! It drops to Z (first layer height) and then moves to the first starting point.
I made my post processing script (https://github.com/foreachthing/Slic3rPostProcessing) just to avoid that (it changes the GCode to move to the first XY first, then drop to Z)!
What you can do, if you don't want/can't use a post processing script, is to move the head to the center of the plate (last line in Start G-Code) like so `G0 X110 Y110 Z10 F12000` (or what fits your needs). Then PrusaSlicer will take over - but it won't crash through the clips of the Ultimaker. We have an UM2 and that one has clips to hold the glass plate... first time using PrusaSlicer - OUCH! Right over a clip!

foreachthing, 2022-03-09:
@fleutot, that blob-thing is to prime the nozzle. That way you really don't need a skirt around your parts - sometimes the skirt can be really short and you still don't have material all the way in the nozzle.

rtyr, 2022-03-09:
I was not aware that this profile needs post-processing to work properly, it needs to be updated.

Minimal amount of filament used for skirt can be specified in print profile.

fleutot, 2022-03-09:
> @fleutot, that blob-thing is to prime the nozzle.

@foreachthing Yes, but the question isn't what it is. It is why my nozzle drags it to the middle of my print, if it (presumably) doesn't for you.

foreachthing, 2022-03-09:
@rtyr: Yeah, this is on me! I didn't think about _not_ using my post processing script! 😵‍💫 
@fleutot because I am using post processing and you don't.

Easiest fix: add `G0 X5 Y110 Z10 F12000` as last line in your Start G-Code. So your nozzle will start somewhere around middle-Y from your bed.
Else, get PS to change how they start the print. I.e. move in XY first, then in Z.

foreachthing, 2022-03-09:
On the right, PS original GCode. Pay close attention to line 54: `G1 Z.2 F9000`. Followed by `G1 X104 Y104` to first skirt point.
[image]

I changed that (on the left) to `G0 X140 Y104 F9000` and after that `G0 F9000 Z3` and then `G0 F3000 Z0.2` to first skirt point.

fleutot, 2022-03-09:
@foreachthing I'm considering changing the start gcode to draw a prime line at Z == 0, just like prusaslicer does with the default Prusa profile. I'll watch for the metal clip at origin, thanks for the heads up.

If that works I'll do a pull request, and we can take it from here.

foreachthing, 2022-03-15:
@fleutot 
Please try this new start G-Code below:
```
; # # # # # # START Header
G21 		; metric values
G90 		; absolute positioning
M82 	; set extruder to absolute mode

G28 X0 Y0 Z0	; move X/Y/Z to endstops
G1 X0 Y0 Z30 F15000	; move to start position

; Heat up bed and nozzle
M190 S{first_layer_bed_temperature[0] - 5}	; wait for bed temperature - 5
M140 S[first_layer_bed_temperature]	; continue bed heating
M109 S[first_layer_temperature]	; wait for nozzle temperature

; M92 E492.45     ; BONDTECH Extruder (uncomment if you have one);

G92 E0		; zero the extruded length
G1 F160 E22	; purge nozzle with filament
G92 E0		; zero the extruded length again
G1 F80 E8	; additional priming
G92 E0		; zero the extruded length again

; Gently scrape filament from the nozzle and avoid clips
G1 F600 E-1 ; retract some to relieve pressure - the brim will take care of unretract
G0 F3600 X25 ; move right to clear clip
G0 F3600 X55 Z{first_layer_height*2} ; drop tool and then ...
G0 F1200 X60 Z{first_layer_height/2} ; ... scrape filament from nozzle for 5 mm ...
G0 F3600 X65 ; ... scrape!
G0 F3600 Y7 Z{first_layer_height*5} ; go to start point for PrusaSlicer; Y7 you ask? To clear clips, again!
; PrusaSlicer will drop nozzle at this point to the plate at first layer height ({first_layer_height}).
; # # # # # # END Header
```

I made following changes:
1. No need for post-processing (that was my mistake and I am sorry!)
2. Removed one of the redundant M107 fans off
3. Now, gently scraping the filament on the plate (at half the first layer height - I don't like Z=0 very much!). All this happens on Y=0.

This might leave a tiny bit of filament between X65;Y0 and X65;Y7. If your printer works the same way mine does, make sure your print is not covering this area.

Use CraftWare (or something else) to preview complete start G-Code. PS-Viewer unfortunately won't allow you to preview the start G-Code.

fleutot, 2022-03-15:
> @fleutot 
> Please try this new start G-Code below:

Thanks, I will have a look. I'm also doing experiments here, with varying success. 

> Use CraftWare (or something else) to preview complete start G-Code. PS-Viewer unfortunately won't allow you to preview the start G-Code.
> 

My prusaslicer (2.4.0) does show the start gcode in preview.

foreachthing, 2022-03-15:
> My prusaslicer (2.4.0) does show the start gcode in preview.

I guess I need to downgrade the previewer ...

foreachthing, 2022-03-15:
I just tried 2.4 but it won't show the moves in start g-code ... what are you doing for the preview to work?

rtyr, 2022-03-15:
You have to check "Travels" in the preview.

foreachthing, 2022-03-15:
> You have to check "Travels" in the preview.

Holly 💩 ! I didn't know that (obviously)!! THANKS!

rtyr, 2022-03-24:
This issue should be solved by last configuration update. We also improved the preview to prevent issues in the preview (it is change for PS2.5.0). Closing.

closed 2022-03-24 by rtyr