prusa3d/PrusaSlicer-settings #155: Raise 5cm at the end of a print

wavexx, 2022-03-30:
In the MKx End G-Code section we have:

``{if max_layer_z < max_print_height}G1 Z{z_offset+min(max_layer_z+30, max_print_height)}{endif} ;``

to raise the extruder 3cm before parking. Can we increase this to at least 5cm? I find the 3cm insufficient most of the time to remove the steel sheet, especially on narrow prints. Going through the Z encoder every time is annoying. 5cm changes very little and is still very conservative, but seems to be sufficient most of the time for me.

We have multiple request for this behavior on the MK3 FW: https://github.com/prusa3d/Prusa-Firmware/issues/2830 https://github.com/prusa3d/Prusa-Firmware/issues/3119

rtyr, 2022-03-30:
I am not aware of any requests regarding this (and those mentioned are mostly related to aborted prints), but I guess we can increase it anyway.

rtyr, 2022-04-27:
Increased to total 50mm in the latest update (1.4.5).

closed 2022-04-27 by rtyr