prusa3d/PrusaSlicer-settings #156: Perform final wipe at the end of a print

wavexx, 2022-03-30:
I've been using this trick for a long time on my MK3: just at the start of the end g-code sequence, perform a quick wipe:

```
M204 S[machine_max_acceleration_y] ; max acceleration
G91 ; relative move
G1 Y-3 F10000 ; quick wipe
G90 ; back to absolute
M204 S300 ; set back average accelleration
M104 S0 ; turn off temperature
M140 S0 ; turn off heatbed
...
```

Moving to Y-3 is always possible due to the space for the purge line. This removes the tiny hair remaining at the top of the print in *most* cases (not all, since this is not a true wipe).

rtyr, 2022-03-30:
I will likely change it in a similar way to current Prusa MINI behavior. (Print end -> only small Z lift -> travel to the XY parking position -> rest of the Z travel). The long Z travel right after the print seems to be the main culprit.

wavexx, 2022-03-30:
I remember trying this in the beginning (along with no Z travel at all). The fast short move is necessary to make a good surface.

If there's still a small zit, it's actually better to leave the long tail visible IMHO, so that I can more easily see where I need to cut it off :)

rtyr, 2022-03-30:
There is always some Z travel before end g-code, unless z-lift/z-hop is completely disabled in printer settings.

wavexx, 2022-03-30:
Heh indeed, forgot about that. I've been disabling z-hop for most of my profiles indeed.
I really like the final quality with the wipe though.
Could you try with z-hop disbled and give it a shot nonetheless?

I think it's worth either undoing the last z-lift, which is kind of ugly, or perhaps simply removing the last z-lift during gcode generation directly in prusaslicer to make this work nicely.

rtyr, 2023-11-21:
Current solution is considered to fine. Closing.

closed 2023-11-21 by rtyr