prusa3d/PrusaSlicer-settings #159: Feature Request: Add profiles for PolyMaker filaments

theacodes, 2022-05-20:
Hello! It would be useful to add filament profiles for PolyMaker's filaments, especially the popular PolyTerra PLA and PolyLite PLA & PETG. It looks like a user on prusaprinters.org has published a [handful](https://www.printables.com/model/134935-polymakertm-filament-profiles).

I'm happy to contribute these myself, but I wanted to check if there's a specific process needed to calibrate the profiles. I have *lots* of PolyMaker filament around, but I only have a Prusa MINI+.

rtyr, 2022-05-20:
I am already in touch with Polymaker regarding adding their profiles, so Polymaker profiles will be likely added relatively soon.

theacodes, 2022-05-20:
Excellent! Let me know if there's anything I can do to help, such as
testing the profiles. 😎

vtjballeng, 2022-10-03:
Another vote for this @rtyr . PA6-CF, PA6-GF are by far the most important to me.

hreality, 2022-11-03:
Id also like to see this happen for their basic and exotics.. I havent quite figured out how to work them into the defaults selection process and importing from their page tries to overwrite everything.  Still workin on how to deal with the software much less overwrites