prusa3d/PrusaSlicer-settings #162: Elegoo Neptune printer profiles

mlee12382, 2022-07-21:
I would like to submit profiles for the Elegoo Neptune line of printers. github isn't allowing me to add the ini file to the issue but all the files are available here: https://github.com/mlee12382/prusaslicer-elegoo-profiles thumbnails and bed model and texture files are also there.

rtyr, 2022-08-02:
Thanks I will look into it.

rtyr, 2022-08-10:
Profiles for elegoo printers are now added to the master with these changes:

- reworked all thumbnails
- I noticed that build volumes are not correct. It was possibly made like that by design because of some clips, but I don't see it as correct solution. So it is now changed to official values.
- removed texture and model for now. It was visible only for 2S, and txt didn't correspond with the bed model/build volume. Also we prefer not to use such big complex "bed" models. I will likely create some basic bed/texture later. We may mark position of the clips on the bed texture.

Profiles will be included in the next 2.5.0-alpha or beta.

closed 2022-08-10 by rtyr

mlee12382, 2022-08-10:
Awesome, thanks, a couple of notes, the Neptune 1 is a smaller build plate and the X and 3 are both taller. And only the 2 and 2D have bed clips if you mark position in the future the 2S, X and 3 have magnetic build plates from the factory  :)
Thanks for helping out with this!

rtyr, 2022-08-10:
What is the maximum build XY volume on stock 2S, X, 3? Official website states 220x220. But I can see on your github repository, you are mentioning it is possible to print 235x235. Does stock firmware allow to go beyond official limits? Interestingly enough, Creality Ender 3 had also similar weird 220 oficial / 235 real limit :).

mlee12382, 2022-08-10:
Everything except the smaller Neptune 1 is 235x235 on the build plate or
220x220 nominal, 2 /2S /2D are 250mm nominal  260 max Z, X is 300 Z and 3
is 280 Z not sure what the max limits on X and 3 are. The Neptune 1 is
210x210

mlee12382, 2022-08-10:
Basically everything other than the values in *common* are correct and the common size just needs to be updated to 235x235 on the bed, then the override  on bed shape can be removed except for the Neptune 1

rtyr, 2022-08-11:
> Basically everything other than the values in _common_ are correct and the common size just needs to be updated to 235x235 on the bed, then the override on bed shape can be removed except for the Neptune 1

OK, updated.

mlee12382, 2022-08-11:
Awesome thanks for your help :) any ideas on when the next version release might be? I had a couple people in the Neptune community that were curious.

rtyr, 2022-08-11:
Soon. Next release (likely 2.5.0-beta) will be probably out next week.

mlee12382, 2022-08-11:
Awesome  :) thanks again!