prusa3d/PrusaSlicer-settings #164: Feature Request: Add 0.6 and 0.8 nozzle profile for Anycubic Kossel Printers

jamilbn, 2022-08-12:
The print quality that I'm getting from Prusa Slicer in my Anycubic Kossel Linear Plus is so much better than the print quality I'm getting on CURA. But Cura calculate all the line sizes for change in nozzle diameter. I would like to use multiple nozzle diameters in Prusa Slicer too.

rtyr, 2022-08-12:
Do you have such profiles? Profiles for 3rd party printers are usually provided by the community.

jamilbn, 2022-08-12:
I don't have them. Who created the profile for my printer did a great job. I get print quality that I never got with any other software. Tried to create a profile for 0,6 but no sucess yet. If I have them working I'll submit them here. Thanks.

closed 2022-08-12 by jamilbn