prusa3d/PrusaSlicer-settings #165: Adding VOXELPLA material to the next prusaslicer build

VOXELPLA, 2022-08-19:
I am looking to add a profile for our new filament, VOXELPLA, to the next build of prusaslicer. How can I do this?

rtyr, 2022-08-19:
You can attach exported profile here (zip), I will check it.

VOXELPLA, 2022-08-19:
Got it; thank you so much. I have attached two presets for our filament, one for Creality and Anycubic, and one general one with our recommended Temp. Let me know if there is any problem.

[PrusaSlicer_config_bundle VOXELPLA PLUS.zip](https://github.com/prusa3d/PrusaSlicer-settings/files/9384108/PrusaSlicer_config_bundle.VOXELPLA.PLUS.zip)

rtyr, 2022-08-22:
What is brand or manufacturer of this filament (we need some vendor for filament wizard)?

VOXELPLA, 2022-08-22:
The brand and manufacturer is VOXELPLA; all of the filament is PLA 1.75mm, and we offer 5 colors, all included in the zip file above. Let me know if there is anything else we can assist you with.

rtyr, 2022-08-25:
I will add the profile to the Anycubic and Creality bundle. We don't have general filament library yet, so it will be available only for Anycubic and Creality printers.

VOXELPLA, 2022-08-27:
Got it. Thank you so much; we are currently testing the Prusa, Ultimaker, and artillery profiles and should have them ready in 2 weeks. When it is ready, should I just add it to this case or open a new issue? Again, really thank you for your help.

owengaspard, 2022-09-11:
@VOXELPLA per request of @rtyr, I have added your profiles for Creality and Anycubic to my PR. Please see https://github.com/prusa3d/PrusaSlicer/pull/8817.

rtyr, 2022-09-12:
Anycubic update was already released last week.

rtyr, 2022-09-20:
Included in the Creality 0.2.2 configuration update. Closing.

closed 2022-09-20 by rtyr