prusa3d/PrusaSlicer-settings #179: New Elegoo Neptune Printer Profiles

mlee12382, 2023-01-02:


rtyr, 2023-01-06:
You can either create a pull request or just attach modified ini/resources here.

mlee12382, 2023-01-06:
> You can either create a pull request or just attach modified ini/resources here.

Alright I will work on that this weekend :) thanks.

mlee12382, 2023-01-07:
@rtyr Here is the updated ini file as a txt since github doesn't allow .ini files. I wasn't sure on the Pro / Plus / Max what to do in the notes especially since they are no longer bowden. I wasn't sure what keywords to use / change or if it even matters? I also noticed all of the models are tagged as PRINTER_MODEL_NEPTUNE2D, not sure if that was intentional or an oversight when setting it up initially? I left them as is in case changing them messes up other stuff.
[Elegoo.txt](https://github.com/prusa3d/PrusaSlicer-settings/files/10365266/Elegoo.txt)

rtyr, 2023-01-17:
> what to do in the notes especially since they are no longer bowden. I wasn't sure what keywords to use / change or if it even matters?

It doesn't matter if it's not used in any condition. I removed it completely.

> I also noticed all of the models are tagged as PRINTER_MODEL_NEPTUNE2D, not sure if that was intentional or an oversight when setting it up.

I changed it according to particular model, but as far as I can see, this keyword is not used anywhere anyway.

Here is the final bundle (it will be compatible with 2.6.0-alpha1 and newer.
[1.0.2.zip](https://github.com/prusa3d/PrusaSlicer-settings/files/10433627/1.0.2.zip)

closed 2023-01-17 by rtyr