prusa3d/PrusaSlicer-settings #184: missing/empty (first) layer

DanBowk, 2023-02-23:
I get an "empty layer between 0 and .5" when using any variant of 2.6. This is with downloaded models and ones I've created.
[image]

rtyr, 2023-02-23:
Please provide your 3MF project file.

DanBowk, 2023-02-24:
> Please provide your 3MF project file.

Where do I add it? I've tried adding it in teh comments but it says file type not supported! It is a .3mf file.

Sorry, complete newbie to github.

rtyr, 2023-02-24:
Archive it to the zip first.

DanBowk, 2023-02-24:
[AK_Bat_helm_4.0_Back.zip](https://github.com/prusa3d/PrusaSlicer-settings/files/10824343/AK_Bat_helm_4.0_Back.zip)

rtyr, 2023-10-27:
The model have a lot of errors. The model needs to be repaired first.

closed 2023-10-27 by rtyr