prusa3d/PrusaSlicer-settings #191: Prusa MK3s 0.25 nozzle PETG temperature

lukasradek, 2023-03-29:
I have been happy printing Prusament PETG with base profiles since 2020 with great results.
Now I came back to PETG after about a year and done some successful prints with 0.4 nozzle and then switched to 0.25 which and got huge problems with adhesion even on large surface areas. Since I haven't changed anything in my settings / printer I assumed some "hardware misalignement" after the nozzle change. After about 4 failures I managed to make the first layer somewhat stick (few small imperfections but nothing compared to just tearing off previously) - albeit it requried vigorous steel sheet cleaning and ridiculous first layer squish. But even then the second/third layer just refused to stick altogether. So I checked the temps and was surprised that I am printing PETG at 230°C.

And so... 230°C in Prusament PETG for MMU is just unsufficient. Looks like you can get away with it with 0.4 nozzle but 0.25 just does not work.

I have seen it discussed here and one other place at it was sweeped under the rug with explanation that it increases MMU reliability.
At least in my case MMU is just fine with 250°C and even if the reliability is higher at 230°C there is not really a point to it if the rest of the printed "doesn't work".

rtyr, 2023-03-29:
Please provide some example project file which you are unable to print with default temps.

lukasradek, 2023-04-23:
Literally any that I try.

[For example this one.](https://github.com/prusa3d/PrusaSlicer-settings/files/11304109/petg-mmu-temp-issue.zip)

rtyr, 2023-04-28:
I was not able to reproduce the issue you are describing across various Prusament PETG colors and printers.

[image]

[image]

lukasradek, 2023-04-28:
Just to be sure... is that a 0.25 nozzle?
If so, that really baffles me what is wrong at my end.

rtyr, 2023-04-28:
Sure, 0.25mm nozzle, 0.15 layer height. It may be partially clogged nozzle or some thermistor issue. There is also quite big tolerance in thermistor reading (2-3%). We may increase temp a little bit, but it seems the main issue is elsewhere.

closed 2023-10-27 by rtyr