prusa3d/PrusaSlicer-settings #210: [ENHANCEMENT REQUEST with code solution delivered as well] Random positioning of initial purge line

m-cas, 2023-10-04:
**Is your feature request related to a problem? Please describe.**
Having the purge line at the same place managed to damage on the long run my original mk3 pei sheet and I want to avoid this in my new MK4.

**Describe the solution you'd like**
Select randomly where to start the purge line so that no damage after repeated purging is happening.

**Describe how it would work**
The G-code for positioning the line randomly select an X component in the first half of the printing table.
Here is the working code I'd love to see making it into production.

```
;
; Extrude purge line
;
G92 E0 ; reset extruder position
G1 E{(filament_type[0] == "FLEX" ? 4 : 2)} F2400 ; deretraction after the initial one before nozzle cleaning

; Original Prusa code commented out
;G0 E7 X15 Z0.2 F500 ; purge
;G0 X25 E4 F500 ; purge
;G0 X35 E4 F650 ; purge
;G0 X45 E4 F800 ; purge
;G0 X{45 + 3} Z{0.05} F{8000} ; wipe, move close to the bed
;G0 X{45 + 3 * 2} Z0.2 F{8000} ; wipe, move quickly away from the bed

; Random start of purge line inside the first half of the plate
{
local rndpos = int(random(15, print_bed_max[0] / 2));
"M117 Purge begins at " + rndpos + "; share the new start\n";
"G0 E7 X" + rndpos + " Z0.2 F500 ; purge\n";
rndpos = rndpos + 10;
"G0 X" + rndpos + " E4 F500 ; purge\n";
rndpos = rndpos + 10;
"G0 X" + rndpos + " E4 F650 ; purge\n";
rndpos = rndpos + 10;
"G0 X" + rndpos + " E4 F800 ; purge\n";
"G0 X" + (rndpos+3) + " Z0.05 F8000 ; wipe, move close to the bed\n";
"G0 X" + (rndpos+6) + " Z0.2 F8000 ; wipe, move quickly away from the bed\n";
}
```

**Describe alternatives you've considered**
none, it works rather well.

**Additional context**
none. It work.

m-cas, 2023-10-04:
Thanks for the feedback. Have you looked at what I am changing exactly? From your comment it sounds a deeper thing than in reality! All stays the same, it is just a random start position (instead of starting always at 15) along *the same line as today* to avoid repeated stress damage to the same point on the printing table.

I am running it on my MK4 without any FW change. It is only a slightly changed version of the custom pre-print G-code; I fail to see any reason for not adopting it, no complex code, no risks, no FW change needed?!?

rtyr, 2023-10-05:
> Have you looked at what I am changing exactly?

Yes, I understand it well enough.

> no risks,

It's rarely the case in reality. For example, your change would make all new project files shared over Printables etc. incompatible with previous slicer versions. We also may change it completely anyway because of MMU3.

I personally don't have strong opinion about this. I think that current state is ok, it is printed outside of print area and users know where exactly it will be. It is also good position for easy and clean removal. From my experience, it is better to have consistent behavior instead of some random factor, which may confuse some users.

m-cas, 2023-10-05:
Could you please explain what would make it incompatible? No sarcasm, I am really interested.

And here I do not agree: it is NOT ok. Using the same spot over and over creates a trench in the plate. It took about 1 year for my old mk3 to develop the problem and I ended up with a pretty good plate but with a little trench where the purge line was repeatedly put. 

I do not see what confusion it creates: you look where the purge line is and you take it away.

Hey you guys are the owner of this thing, if you want to keep it like this, be my guest :D I am just a user reporting a problem and giving a solution for once.

rtyr, 2023-10-05:
> Could you please explain what would make it incompatible?

Unsupported variables, you will get parsing error with <PS2.6.0.

m-cas, 2023-10-05:
is there a way to test for the slicer version in macro?

harriettubgirl, 2023-10-11:
This is a great suggestion, I can definitely see my smooth sheet wearing out where the purge line is always laid. Please give this as an option!

m-cas, 2023-10-18:
@rtyr just a comment on breaking compatibility, and do not take this as polemic, just the opinion of someone that has been writing software for nearly 40 years. This is the normal state when you release new features; now that you have variables and macros you cannot say "do not use them 'cause they break compatibility with older releases". What is the point of releasing the new stuff then?

m1bcodes, 2023-11-19:
I agree that this would be great feature. 
The structured print sheet gets worn out and the purge line very hard to remove.

m-cas, 2024-11-06:
Now that more than a year has passed and probably nobody uses any slicer below v2.6 (removing the objection mentioned above), could we please have this added to the code?

SPooks753, 2025-12-06:
Yes, please. My printbed has litteraly a hole in shape of a line where the purge is done. It's getting almost impossible to remove it after a print. Even when adding tape on the plate before.