prusa3d/PrusaSlicer-settings #212: Missing Prusament profiles for Prusa XL - 5T 0.6 nozzle?

ulab, 2023-10-11:
With the latest configuration update I seem to have "lost" a lot of Prusament profiles for my XL 5 tool with 0.6 mm nozzle.

While I have selected them:

[image]

They do not show up in the selection box anymore?

[image]

Am I just too tired to see them? ;-)

ulab, 2023-10-11:
Update: I have deleted the PrusaSlicer Config folder, restarted PrusaSlicer, added the XL 5T 0.6 and some Prusament profiles.

They show up now with the old default config:

[image]

I then did `Check for Configuration Updates` and upgraded to 1.0.4 (Templates) and 1.10.1 (Prusa Research). After that they were gone:

[image]

ulab, 2023-10-11:
Ok, this is getting weird. 

It seems like the profiles got unselected in the Configuration assistant too and they are back after re-selecting them.

That never happened during a Configuration update?

rtyr, 2023-10-12:
Previously all filament profiles were shared between XL and MK4 printers. It changed since configuration update 1.10.0, where we internally split XL and MK4 filament profiles (XL needs some special parameters for new ramming feature). That is the reason why XL users had to re-enable some filament profiles in the configuration wizard. 

It was mentioned in release note for 1.10.0 update. You might missed this warning, if you updated straight to 1.10.1.

ulab, 2023-10-12:
That might be the reason. Especially when you start from scratch like I did as a test you'll skip the release notes from 1.10.0. 

It will fix automatically with the next PS release which should come with the new profiles from the start, so I'll just close this issue.

closed 2023-10-12 by ulab

ulab, 2023-10-12:
PS: [1.10.0 Release Notes](https://files.prusa3d.com/?latest=slicer-profiles&lng=en_US) didn't really give a "warning" on this:

> - Enabled XL ramming feature for most filament types (all except PLA and flexibles). This feature reduces strings on the prime/wipe tower (Prusa XL Multi-Tool).
> - Added new profiles for Prusa MINI Input Shaper (Alpha). Added profiles for additional nozzle diameters.

rtyr, 2023-10-12:
It did.

This is the notification text, which was visible in the PrusaSlicer. It is non-detailed release note, which is always visible.

`Added new profiles for Prusa MINI Input Shaper (Alpha). Enabled XL ramming feature. Prusa XL users may need to re-enable filament profiles in configuration wizard.`

The text is taken from 
https://github.com/prusa3d/PrusaSlicer-settings/blob/master/live/PrusaResearch/index.idx

rtyr, 2023-10-12:
We will try to improve the wizard, so it behaves better with fresh install.