prusa3d/PrusaSlicer-settings #219: Spectrum PETG HT100 incorrect temperature during nozzle cleaning

nazar-pc, 2023-11-10:
### Description of the bug

I've got Spectrum PETG HT100 Clear filament and when printing using PrusaSlicer 2.6.1 on Prusa XL it was using temperature 195C when cleaning the nozzle, which is way too low, filament was solid, temperature range written on the spool is 250-280C.

I set the temperature manually to 250C and nozzle cleaning succeeded.

### Project file & How to reproduce

1) Have Prusa XL
2) Have PrusaSlicer 2.6.1
3) Select Spectrum PETG HT100 filament
4) Try to print something with it

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.1+linux-x64-GTK3

### Operating system

Linux

### Printer model

Prusa XL

nazar-pc, 2023-11-11:
I think more tweaks are needed for this filament.

With 250C adhesion with printbed was not enough when using 0.2mm nozzle, it needs to be at least 255C (ideally 260C) for nozzle cleaning to work on Prusa XL.

As for the print, it worked fine with stock 0.6mm nozzle on XL (for low height prints at least), but with 0.2mm nozzle I was getting layer separation.

To fix that I decreased fan speed from 100% to 60% and increased nozzle temperature from 250C to 265C (well within 250-280C range written on the spool).

Would be nice if someone can tweak the profile to take those into account. I see where some of the values are, but not others (like where 195C comes from).

rtyr, 2023-11-11:
Nozzle cleaning / MBL temperature is based on filament type (condition in the start g-code), it's not defined by filament profile. The temperature is typically significantly lower than printing temperature, except for materials like PC, Nylon. This filament may need special treatment outside of "PET" category. We will think about possible solutions with such cases.

Filament profile for our printers was provided directly by filament manufacturer, which doesn't mean it cannot be improved (with mostly used nozzle diameters in mind). Thanks for your input.

nazar-pc, 2023-11-11:
Now it makes more sense, thanks!

Looks like HT prefix in the name stands for "high temperature", so while PETG, it is a bit of a special kind indeed. From my observation it is more or less solid until about 230 degrees.

rtyr, 2023-11-17:
The latest configuration update 1.11.3 for 2.7.0-rc1 should fix the issues mentioned here. MBL is performed at 250C for this filament, temp for 0.25-0.5 was increased to 260C, max fan speed decreased to 65%.

closed 2023-11-17 by rtyr