prusa3d/PrusaSlicer-settings #223: Prusa Slicer error when saving gcode files due to new .bgcode binary half default

dimitarsp, 2023-12-05:
### Description of the bug

The new version of prusa slicer defaults to saving files to .bgcode, but by default saving to .bgcode is disabled, leading to almost always getting an error, just by doing what I was doing before, exporting to flash drive.

Now, instead of simply exporting, i hit export, input a name, hit save, and i get the error. And then i go again, and have to consciously select the third item in the list, .gcode.

Somehow with this bug the time to process 3d prints has significantly extended. 

It would be best to set the default of saving back to .gcode or to give us an option to choose what the default export file type should be. 

### Project file & How to reproduce

Open a new file, import stl, slice it, and export to computer or disk. Type in a name, and you get an error, because .bgcode is not enabled in the settings. 

[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.0

### Operating system

Windows 11

### Printer model

prusa mini+

rtyr, 2023-12-05:
I checked and you are right, even though this issue is present only in "old" non input shaper profiles for 0.6 nozzle. We will fix that with configuration update. Thank you for reporting it.

rtyr, 2023-12-05:
If you have up-to-date firmware (5.1.0), you can enable "MINI Input Shaper" through configuration wizard.

rtyr, 2023-12-05:
Fixed with configuration update 1.11.6.

closed 2023-12-05 by rtyr