prusa3d/PrusaSlicer-settings #227: Elegoo Neptune 4

Chr1s70ph, 2024-01-02:
There has already been an Issue repo, but there the issue was closed, so I'm bringing this here, since I would like to have support for my Neptune 4 in PrusaSlicer.

https://github.com/prusa3d/PrusaSlicer/issues/11087

> I'm not the owner of the original issue, that was @luisillo26

rtyr, 2024-01-02:
As neophyl already mentioned there, 3rd party profiles are usually provided and maintained by the community.

Chr1s70ph, 2024-01-02:
> As neophyl already mentioned there, 3rd party profiles are usually provided and maintained by the community.

Just curious, since Neptune 1-3 are all in here, how does the community "provide and maintain" the profiles?
Does this work via Pull Requests?

rtyr, 2024-01-03:
Yes.

https://github.com/prusa3d/PrusaSlicer/wiki/Vendor-bundles-and-updating-process

iuvi7, 2025-04-10:
So, i can just use 3Pro for my 4Pro?