prusa3d/PrusaSlicer-settings #228: [BUG] A number of MK4/XL Filament Profiles inherit the IS temp settings rather than the normal temp settings regardless of printer IS status

IBNobody, 2024-01-05:
I found that when not using IS, the default PLA temperatures for Overture, eSun, Hatchbox, and many other PLA filament vendors default to the IS temps rather than the normal temps.

This causes confusion. 

For example, if I have a model I am printing using Overture PLA on my MK3 and transfer it over to my MK4 (non-IS), the temperatures will jump from 215/210 to 230/220.

(Not every vendor is set up like this. Obviously Prusament is set up with individual non-IS and IS profiles. 3D-Fuel is also good.)

rtyr, 2024-01-05:
Expected behavior. It is not inherited, it just uses the same shared filament profiles in these cases. Only IS printer profiles will remain in the future and old non-IS printer variants will be soon removed completely. Both variants were available only as for some transition period.

IBNobody, 2024-01-05:
So to clarify, in the future, there will no longer be any non-IS profiles (filament, print, printer) for the MK4 and XL?

If that is the case, I appreciate having fewer profiles to manage.

rtyr, 2024-01-05:
Yes, we are planning to remove all those non-IS printer variants from the slicer. XL, MK4 and MINI are using IS FW by default for some time. MK3.9 is like that from the beginning, because nonIS version of that printer never existed.

IBNobody, 2024-01-05:
Never have I loved to close a "working as intended" ticket with more gusto. Thank you very much.

closed 2024-01-05 by IBNobody