prusa3d/PrusaSlicer-settings #237: Add Extrudr PETG filament to Prusa i3 MK3S + MMU2S

chrisboer, 2024-02-21:
### Description of the bug

I can not add Etrudr Petg filament as the PrusaSlicer states that this is not compatible with my Prusa i3 printer, I would not know why ?

[image]

### Project file & How to reproduce

1. Open prusa slicer
2. GO to Filament selection
3. Add other filaments
4. Select All printers (on there are only  Prusa i3 MK3S + MMU2S and  Prusa i3 MK3S + MMU2S SIngle 
5. Then the above screenprint can be seen and no Extrudr PETG filament can be added

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.1

### Operating system

Windows 11

### Printer model

Prusa i3 MK3S + MMU2S

rtyr, 2024-02-21:
Only selected filament profiles are compatible with MMU mode. Also such profiles have different parameters than same filaments in "single" mode (lower temperature in general). You can try to use "Generic PETG" profile instead.

chrisboer, 2024-02-21:
What has an MMU to do with temperatures this is only the multi filament selector.  It is selectable in single MMU mode. So if it needs to change from fila1 to fila2 it should only impact the hotend ? So I am totally confused with the logic behind this answer.

rtyr, 2024-02-21:
There are no toolchanges in the single mode (if you print only with one filament - "MMU2S single" printer). If you print multimaterial print ("MMU2S" profile), lower temps are usually needed for more reliable toolchanges (basically the tips ale less stringy after unload from the hotend).

Please contact our tech support if you need further info.

closed 2024-02-21 by rtyr