prusa3d/PrusaSlicer-settings #239: Filament specific G-code vs. nozzle sizes on the XL

smurfix, 2024-02-26:
### Description of the bug

The filament g-code contains code like `{if nozzle_diameter[0]==0.4}`.

The problem is that this refers to the first nozzle. What if this filament is on the second or fifth extruder, with a different nozzle size? In that case the K values are wrong and the print is ugly.

### Project file & How to reproduce

untested because I don't want to waste filament, but looking at the code this seems to be what happens.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.2rc1

### Operating system

Linux Debian

### Printer model

XL

rtyr, 2024-02-26:
> What if this filament is on the second or fifth extruder, with a different nozzle size?

There is no such variant within bundled profiles (one printer profile with different nozzle diameters). This is generally not supported properly by PrusaSlicer at the moment. We may change it to [filament_extruder_id] in the future.

rtyr, 2024-04-25:
The condition was changed to `nozzle_diameter[filament_extruder_id]`.

closed 2024-04-25 by rtyr