prusa3d/PrusaSlicer-settings #243: Lower the XL bed leveling probe temperature for FLEX from 210 to 175

ssill2, 2024-03-02:
The default printer start gcode for the XL 5T has this line
```
M109 T{initial_tool} S{((filament_notes[initial_tool]=~/.*HT_MBL10.*/) ? (first_layer_temperature[initial_tool] - 10) : (filament_type[initial_tool] == "PC" or filament_type[initial_tool] == "PA") ? (first_layer_temperature[initial_tool] - 25) : (filament_type[initial_tool] == "FLEX") ? 210 : (filament_type[initial_tool]=~/.*PET.*/) ? 175 : 170)} ; wait for temp
```
The 210 temperature set for flex is way too hot and causes small dots of filament to be left over the entire print area right before the print starts.  These are super annoying to remove.   This happens for both ninjaflex and sainsmart tpu and I suspect others as well.   Would it be possible to make 175 the default for flex bed leveling probe temperature?  Or is there a reason 210 was chosen?

rtyr, 2024-03-02:
210C was found to be most reliable temp for flexible materials regarding loadcell/probing accuracy (which is the main priority).

ssill2, 2024-03-02:
For which brand of flex?
For sainsmart and Ninjaflex i just get a bed full of dots like this

https://github.com/prusa3d/PrusaSlicer-settings/assets/25040169/fe2576b5-9ad2-47fe-a77a-0713109498b1

My work around of 175 works for the two brands of flex I typically print with but I have to remember to go back and make the change each time a create a project or if the settings bundle is updated here.   I've printed quite a lot of flex since I got the xl and since I defaulted to 175 without any issue.  Is there any chance of lowering the 210 or is that set in stone?

rtyr, 2024-03-02:
We did a lot of testing during the development and lower temps showed unreliable results with flexibles in general. Dots on the bed are unfortunately a possible side effect.

It's not so much slicer/settings topic as firmware/hardware topic.

ssill2, 2024-03-02:
I'm curious what kind of unreliable results you were getting during testing.  Adhesion issues?   Did the team see these dots?  I was one of the early 5T preorders so mine originally came with .06 nozzle, which is what the above was taken with.  I've since replaced them with .4mm nozzle and while it still would do it for 210, it's not nearly as bad.   There's a lot of room between 175 and 210, perhaps there's a happy medium.

rtyr, 2024-03-02:
Basically bad first layer (incorrect z offset). The "dots" can happen with materials, which require higher probing temp or with materials with high level of moisture.

Anyway, I will pass this issue to the Firmware/Testing teams.

ssill2, 2024-03-02:
awesome thank you.  I know others on the forum are having the same issue.

closed 2024-03-02 by ssill2