prusa3d/PrusaSlicer-settings #247: Walls not stick with each-other (separated) using stock Prusament filament and profiles on Prusa Mini IS

microjean, 2024-03-25:
Walls do not stick with each-other (separated) using Prusament PETG filament and unchanged PETG profiles on Prusa Mini IS. Prusa slicer 2.72, latest firmware for Prusa Mini.

rtyr, 2024-03-25:
Please provide your 3MF project file.

microjean, 2024-03-25:
I attached a 3mf project file.
[xl4015_enclosure.zip](https://github.com/prusa3d/PrusaSlicer-settings/files/14743467/xl4015_enclosure.zip)

rtyr, 2024-03-26:
Thanks, this issue is mainly related to Arachne perimeter generator, which produces too wide extrusions in some cases. We are planning to improve this on the slicer side.

I recommend to switch perimeter generator to "classic" for this particular object.