prusa3d/PrusaSlicer-settings #249: MK3.5 senseless start heat up of the nozzle to print temperature

woecj, 2024-04-09:
Since I updated my MK3s+ to MK3.5, I see the following print startup procedure:

1. heat up nozzle to print temperature
2. cool down the nozzle to 170°C
3. make the PINDA measurements
4. heat up nozzle to print temperature
5. start printing

I cannot find a reason for step 1 - it only leads to filament leakage while waiting ...
Please check the slicer software (I use the latest version 2.7.4)
Also, I noticed that the default G-code type marlin 2 is not the type mentioned in the hoover message to be standard (RepRap/Sprinter) - which one should be used?

rtyr, 2024-04-09:
Are you sure you are using stock printer profile? There is no such behavior in stock start g-code.

"Hoover message" (tooltip) shows application defaults, not printer default.

If you think you are using the correct printer profile, please provide your 3MF project file.

rtyr, 2024-04-09:
Also do you have MMU unit installed on your printer?

woecj, 2024-04-09:
No MMU installed.
I checked the profile setting: I used an original 3.5 setting migrated  from MKS3s+. When I change to a original 3.5 setting, the start code is different. I will check if this change the behavior.

rtyr, 2024-04-09:
>  I used an original 3.5 setting migrated from MKS3s+
I am not sure what do you mean by this, but I guess you likely transfered start g-code from different printer profile (or from 3MF project with different printer).

rtyr, 2024-04-09:
I am closing this issue, the gcode is ok, but feel free to respond here if you think there is some problem.

closed 2024-04-09 by rtyr