prusa3d/PrusaSlicer-settings #251: add the 0.30 Printer settings profile

cgerrist, 2024-04-21:
Hello,

In the mk3s 0.40 printer settings I had the 0.30 Printer Settings profile.
This was an perfect setting for speed and quality.
Could this profile add in the next release?

Chris Gerrist

cgerrist, 2024-04-21:
I forgot to write my printer version. I have a mk3.5 printer

rtyr, 2024-04-23:
It made sense on MK3S, but not so much on MK3.5, since even 0.20 SPEED already prints so fast, that it is close to the volumetric speed limits of the hotend (how fast can be filament melted). 0.3 layer height would need to be printed much slower.

cgerrist, 2024-04-28:
Hi,

Today I made a print profile 0.30 for the 0.4 nozzle. Same as the Prusa mk3s profile.
The print looks beautiful. I didn’t had any problems at all.

Chris Gerrist

rtyr, 2024-04-28:
It will print fine, but slower. The speed is limited automatically by volumetric speed limit.

rtyr, 2024-08-29:
more layer heights were added for high-flow nozzle profile variants

closed 2024-08-29 by rtyr