prusa3d/PrusaSlicer-settings #253: Please reduce the PCBlend temp for bed leveling to avoid nozzle cleaning failures

m-cas, 2024-04-23:
Unlike PETG and PLA, every single time I try to print with PC Blend the nozzle cleaning fails. The reason is that the nozzle temp is so high that a constant stream of material oozes through it and prevents a successful cleaning.

Reducing the temperature of PC blend by 50 degrees instead of the currently implemented 25 makes the problem go away in my case.

I am using an MK4 (kit) inside a Prusa Enclosure.

I changed these two lines in the slicer printer setting and substituted -50 instead of the default -25:

M104 T0 S{((filament_notes[0]=~/.*HT_MBL10.*/) ? (first_layer_temperature[0] - 10) : (filament_type[0] == "PC" or filament_type[0] == "PA") ? (first_layer_temperature[0] - 25) : (filament_type[0] == "FLEX") ? 210 : (filament_type[0]=~/.*PET.*/) ? 175 : 170)} ; set extruder temp for bed leveling
M109 T0 R{((filament_notes[0]=~/.*HT_MBL10.*/) ? (first_layer_temperature[0] - 10) : (filament_type[0] == "PC" or filament_type[0] == "PA") ? (first_layer_temperature[0] - 25) : (filament_type[0] == "FLEX") ? 210 : (filament_type[0]=~/.*PET.*/) ? 175 : 170)} ; wait for temp

neofir3, 2024-10-26:
Yes i have the same Problem

ryanpriore, 2025-02-11:
I occasionally have the same issue with the MK4(S), but it is very apparent with my CoreONE. I had to drop the temperature by -75. I tried lowering to -50 from the default -25 without success. This is the case for both Prusament PC-blend and generic PC filament.

m-cas, 2025-02-11:
-50 worked for me on my MK4; with my MK4S and the high flow nozzle, I had to go down another 25C as well, so I confirm your finding @ryanpriore !