prusa3d/PrusaSlicer-settings #255: Decrease default bridging speed for Prusa MK4IS 0.6mm presets (and further tuning)

Fox2wo, 2024-05-09:
PrusaSlicer 2.7.4

The default configs for the 0.6mm nozzle for the MK4 have very poor bridging performance when printing with PETG, with very droopy bridges that develop holes and appear under extruded at the start

The 0.4mm nozzle config has a bridging speed of 50mm/s, while the 0.6mm nozzle has a bridging speed of 40mm/s. Based on the area of the nozzle bore, the 0.6mm preset should have ~44% of the bridging speed that the 0.4mm preset has.

This lower speed would allow for better, neater bridges, more comparable to the 0.4mm nozzle.

I feel like the 0.6 config is far less tuned compared to the 0.4 config. Less filaments are supported by default (Tungsten 75% PETG, ColorFabb foaming TPU, etc.), more oozing before probing, leading to inconsistent first layers, as well as more nozzle rubbing on infill causing weaker parts and plastic accumulating on the nozzle.

[image]
^ Bridging with 0.6mm nozzle

rtyr, 2024-05-09:
Please provide your 3MF project file.

Fox2wo, 2024-05-09:
> Please provide your 3MF project file.

https://drive.google.com/file/d/108fXZITWVF87e7Tv_LLNsy-boTR9-KM7/view?usp=sharing