prusa3d/PrusaSlicer-settings #256: Prusament PLA Blend profile uses  nozzle_diameter[0] instead of nozzle_diameter[filament_extruder_id] in the start code

codingcatgirl, 2024-05-20:
This is probably wrong, as it has been changed to nozzle_diameter[filament_extruder_id] in all other profiles. I'm even wondering if it might be the cause for load/unload issues i've been facing.

rtyr, 2024-05-21:
It does not matter at this point. The change was mainly about easier experiments with combining different nozzle diameters within one prints (and it still only affects pressure advance value).

>  I'm even wondering if it might be the cause for load/unload issues i've been facing.

In reality, there is no difference between [0] and [filament_extruder_id] unless you are experimenting with combining various nozzle diameters within one print (and it still does not affect load/unload).

Anyway, thanks, we will change it here as well in some future update.

closed 2024-05-21 by rtyr

rtyr, 2024-05-21:
Actually that filament already uses [filament_extruder_id], so there is nothing to change. You probably meant MK4 variant of that profiles where it does not matter (it always uses only one nozzle diameter)

codingcatgirl, 2024-05-21:
Oh, of course. Makes sense in hindsight, i guess 1 am at night is pretty late :D