prusa3d/PrusaSlicer-settings #257: Rat Rig Profiles Update

nameisCruzCruz, 2024-05-29:
Hi! I'm Cruz from RatRig and would like to submit a RatRig.INI file with updates:

We restructured the file, to be more intuitive to make future changes and find parameters, we used inheritances to make some settings universal like speed, layer height, and nozzle settings, so changing them in one place will directly impact all profiles, avoiding future mistakes.
-  Added the V-Core 4 CoreXY and Hybrid profiles, with 0.4; 0.5 and 0.6 nozzles.
-  Added 0.5 and 0.6 nozzle options to the V-Core 3 and V-minion as well as updated their parameters with new prusa slicer features.
-  Added the Punkfil filament line with their respective settings.

The V-Core 4 has new PNG images but uses the same bed and print surface from v-core3. I've also attached the IDX file with the updated change number 1.1.0.
We were able to test the INI file on many computers and do a lot of print tests to validate our submission.
I hope all files are correct for this submission, if not, I'll make the necessary changes/updates as required. Hope this is the right place for this subject, as mentioned in https://github.com/prusa3d/PrusaSlicer/wiki/Vendor-bundles-and-updating-process#creating-a-profile-for-a-vendor-not-currently-present-in-prusaslicer bullet point 12), it's simpler than making a PR.
Files bellow:
[RatRig Profiles PrusaSlicer 1.1.0.zip](https://github.com/prusa3d/PrusaSlicer-settings/files/15487867/RatRig.Profiles.PrusaSlicer.1.1.0.zip)

Thank you!

rtyr, 2024-05-30:
Hello. I checked the bundle and found some small issues, which needs to be fixed.

 **The preset `RatRig V-Core-3.1-400 0.6Nozzle` inherits an unknown preset "`*V-Core3_0.6Nozzle*`".**

It seems that "`*V-Core3_0.6Nozzle*`" preset is missing completely in your bundle.

**The preset `RatRig V-Minion-180 0.4Nozzle` inherits an unknown preset "`*0.4nozzle*`"
The preset `RatRig V-Minion-180 0.5Nozzle` inherits an unknown preset "`*0.5nozzle*`"
The preset `RatRig V-Minion-180 0.6Nozzle` inherits an unknown preset "`*0.6nozzle*`"**

There are no such presets within printer section.

**Many of print presets uses `variable_layer_height` key, which is not valid key for print preset (it's printer setting, not print setting).**

You can check for errors like this by running prusa-slicer-console with --loglevel=5 parameter.

nameisCruzCruz, 2024-05-31:
Hi!
Rookie mistakes, thanks for pointing them out.
Fixed them:

[RatRig Profiles PrusaSlicer 1.1.0.zip](https://github.com/user-attachments/files/15510027/RatRig.Profiles.PrusaSlicer.1.1.0.zip)

Not to rush anything, just wondering what is the average time until the profiles are available (considering there are no mistakes on this version)
Once again, thank you so much for your time.

rtyr, 2024-05-31:
There were still some issues, but I fixed it.

- invalid key variable_layer_height was still used in all print profiles (it was inherited from common print profille).
- there were some typos in several printer profiles (rinter_variant instead of printer_variant, CHYBRID instead HYBRID)

I fixed the issues and pushed the update, so it is already available for users.

closed 2024-05-31 by rtyr