prusa3d/PrusaSlicer-settings #273: Flashforge Creator Pro II IDEX printer

paynterf, 2024-09-06:
jacotheron and I have created a profile for the Flashforge Creator II Pro IDEX printer, based closely on the BIBO dual-extruder printer.  We started with BIBO.ini and (mostly) just changed all occurrences of 'BIBO' with 'Flashforge Creator Pro 2'.  After placing this in the vendor subdir as 'Flashforge.ini' we were able to use it without problems until the recent update of Prusa Slicer to 2.8.0.

After this update the only way we could get our printer profile back was to downgrade to 2.6.1, get the profile recognized again, and then update to 2.8.0.  However, even now 'Flashforge' isn't shown under 'other FFF' in the config wizard, although all the printer configuration (dual extruder, left only, right only, etc) are available in the drop-down menus in the plater.

I have read and re-read all the documentation for the config files and the accompanying index files, and AFACT, this should all be recognized under 2.8.0, but there is obviously something wrong.  Any ideas why the Flashforge.ini and Flashforge.idx files in the vendor subdir aren't being recognized?

TIA,

Frank
[FlashForge.zip](https://github.com/user-attachments/files/16913026/FlashForge.zip)

rtyr, 2024-09-06:
1) version in your ini is 0.0.1-alpha3, there is no such version in the IDX file
2) "alpha" labeled versions are not accepted by final versions of the slicer
3) missing repo_id in the vendor section

Fixed files are here:
[fix_FlashForge.zip](https://github.com/user-attachments/files/16913180/fix_FlashForge.zip)

paynterf, 2024-09-07:
Thanks!  The 'fixed' files work fine, and I now have the 'Flashforge' choice under 'Other FFF' - yay!  However, when I click on 'FlashForge FFF' I get 'No Image', even though I have a 'CreatorPro2.png' image file in the vendor subdir, and the line

thumbnail = CreatorPro2.png

In the Flashforge.ini file (I also tried 'thumbnail =  CreatorPro2.jpg', as CreatorPro2.jpg was 1.3MB, while CreatorPro2.png was over 10MB)
[image]

Can I buy another clue? :)

Frank

rtyr, 2024-09-12:
You have to create `FlashForge` subdir in the `vendor` directory and place the png file there.
vendor\FlashForge.ini
vendor\Flashforge.idx
vendor\FlashForge\CreatorPro2.png

paynterf, 2024-09-12:
Roman,

Thanks for the information, but even with the CreatorPro2.png in the
FlashForge subdir, it's still not working.  Thoughts?

TIA,

Frank

[image: Screenshot 2024-09-12 105606.png]

[image: Screenshot 2024-09-12 105650.png]

rtyr, 2024-09-12:
The image should have some adequate resolution (it is a thumbnail), standard resolution used by us is 180x256. Looking at your photo attached here, it is way too big :-).

paynterf, 2024-09-13:
I reduced the image size to 148x160 (55KB) and it was still rejected.  Is
there some other restriction as well as size?

[image: Screenshot 2024-09-13 190250.png]
[image: Screenshot 2024-09-13 190231.png]

Frank

rtyr, 2024-09-16:
I don't see any issue, it works fine. Make sure you have correct extension specified, also if you are on Mac or Linux, it is laso case sensitive (PNG vs png is not the same).

Anyway I tried attached files and it works fine.
[FlashForge.zip](https://github.com/user-attachments/files/17010171/FlashForge.zip)

paynterf, 2024-09-16:
> specified, also if you are on Mac or Linux, it is also case sensitive (PNG

Thanks, that worked!

Thanks again for all your help getting my FlashForge CreatorPro2 profile up
and running.  Is there a way of making this available to other Prusa Slicer
users?

Regards,

Frank

rtyr, 2024-09-24:
It would be possible, but I am not happy about the fact it needs additional post-processing.

paynterf, 2024-09-24:
I understand. I'm not happy either, but it seems to be the only way to
convert the .bdcode format output from PrusaSlicer to the .gx format used
by FFCP2.  If you have an alternative, I'd love to hear about it! 😁  The
good news is, the Python script developed by Jacotheron ( GitHub -
Jacotheron/FlashForge-CreatorPro2-PS-Profile: PrusaSlicer Profile for the
FlashForge CreatorPro2 3D printer
<https://github.com/Jacotheron/FlashForge-CreatorPro2-PS-Profile>) seems to
work very well, and except for very large print files isn't noticeable at
all.

Regards,

Frank