prusa3d/PrusaSlicer-settings #277: Ratrig profiles are missing a . in comnpatible printers condition

evilC, 2024-11-01:
All the RR profiles have something like `compatible_printers_condition = printer_notes=~/.*VCore4HYBRIDsettings*/`  
If you add anything (except, in this case, as many `s` as you like...) to the end of a printer's notes, it breaks that setting for that printer.  
The above should be changed to something like `compatible_printers_condition = printer_notes=~/.*VCore4HYBRIDsettings.*/`

rtyr, 2024-11-01:
FYI @nameisCruzCruz

As I understand it, it only affects the case, when someone creates his own custom printer profile and changes printer notes.

nameisCruzCruz, 2024-11-04:
Hi @rtyr printer notes aer meant to automatically select the print profiles as you know, for now it is mandatory on our INI file, maybe I can work around it to prevent the user from breaking it. Do you have any suggestions?