prusa3d/PrusaSlicer-settings #282: Some Prusa Mini+ profiles for filaments use too low fan speeds, leading to print fan is not spinning error

nvtkaszpir, 2024-12-15:
Some Prusa Mini+ profiles for filaments use too low fan speeds, leading to print fan is not spinning errors.
Looks like firmware requires higher speeds, some profiles already have minimum speed of 25%, but 15% is just way too low.

Related https://forum.prusa3d.com/forum/general-discussion-announcements-and-releases/print-fan-is-not-spinning-error/#post-729846

rtyr, 2024-12-15:
FW automatically ramps the speed to the minimal supported speed (which is not visible on LCD), I believe it is around 20% currently. Fan starts to spin at around 3% (but that is already that increased PWM in reality). Your issues are likely caused by incorrect reading of fans rpm, which can be caused by HW (fan) or by some FW issue with rpm reading.

closed 2024-12-15 by rtyr

nvtkaszpir, 2024-12-15:
I'm using Fiberlogy PETG@MINI which sets minimum fan speed to 15%.
Notice that Prusament PETG has it set to 30%, so bumping it in Fiberlogy solved the issue.

The fan spins properly, the low value from filament is still used anyway, so maybe firmware is not really enforcing it?

I did not factory reset after firmware update to Mini 6.1.2, so maybe there is a problem in that?

rtyr, 2024-12-16:
You can try different fan speeds via LCD (without printing). I believe you will see/hear no difference between for example 5%, 10% and 20%. FW may have some issues with reliably reading fan rpm for some reason.